night-ghost/minimosd-extra #130: OSD_config.exe not showing OSD items over panel images

mr-sneezy, 2016-04-09:
Not sure if I have missed something I should have done, but I don't seem to get the OSD overlay preview in the OSD_Config tool panel screens like with the old version tool, maybe it's an OSD_config tool bug. 
I do get OSD items visible via a Pal videop monitor (playing TLOG) from the minimOSD output though. 
So the bug is in OSD_config I think. I get a red position box indicator while adding an item, but nothing after that.
[image]

night-ghost, 2016-04-10:
You don't choose font

closed 2016-04-10 by night-ghost

mr-sneezy, 2016-04-10:
Choose a Font ?
Uh, had no idea it was required sorry.

mr-sneezy, 2016-04-10:
OK, where is this Font set. I can't find help in a readme in the Release folder.

mr-sneezy, 2016-04-10:
OK, I've looked again and cannot see a place in the OSD_Config.exe software for a screen Font. Please tell me where this Font parameter is set now. 
Thanks.
(BTW you are not meaning the Character Set we load to the MAX7456 are you)

night-ghost, 2016-04-10:
This is not the place for individual consultations. It works for all, check your mistakes.

mr-sneezy, 2016-04-10:
Well I never did find the bug, after restarting again I found the problem resolved itself. 
I have other problems now and will discuss elsewhere, or just go back to the old firmware as it was very stable for me.

night-ghost, 2016-04-10:
if you have not even tried to restart the program before you to report bugs, why your opinion should be interesting to me? All your problems from your hands.

mr-sneezy, 2016-04-10:
You jump to conclusions about people very fast my friend don't you. Maybe you are a good mate of TimeCop ?

Yes I rebooted my laptop after downloading the files again, yes I flashed the new firmware again (twice), yes I reloaded the character set twice or maybe three times. I also then downloaded a fresh copy of the whole github repo before I gave up and posted an issue to you here. 

Anyway, I shant bother you again.