night-ghost/minimosd-extra #133: Update of OSD data freezes with PAL video signal from Turnígy IC-120NH

magnus1089, 2016-04-17:
When using version 827 with a PAL video from Turnigy IC-120NH I get no updates on OSD data. When no video signal or signal from GoPro 3 Black (PAL or NTSC) OSD data updates normal. I have similar setup on two quads with same behaviour. Tried both Auto, PAL and NTSC setting. I reverted back to old R800 version.

night-ghost, 2016-04-25:
this is some kind of mysticism, can you show video?

athertop, 2016-04-25:
Could it be that the Stream Rate settings are somehow not getting through from the MinimOSD board to the Ardupilot board? Maybe the Tx from MinimOSD is not connected? Maybe you should manually set the SRn_ values in Mission Planner? just an idea. I just upgraded MinimOSD-Extra to this night-ghost revision and it works flawlessly for me. Very happy with it.

closed 2016-05-31 by night-ghost