night-ghost/minimosd-extra #135: OSD restarts/blanks out on disarming the copter

ivanskj, 2016-04-29:
When disarming after a flight the OSD reboots, and effectively zeroing out meassured mAh, tested on 824. Any other info i can provide to this issue?

zach030801505, 2016-05-02:
I'm having the same problem, I'm just glad it's not happening in flight

tops4u, 2016-05-04:
Same issue here. Either it states "No Data" or just blanks after quickly showing the Resumee. Using 827 with PX4 Stack.

thecodingfrog, 2016-05-05:
Same issue, I'm using only 1 screen. What about you guys ?

tops4u, 2016-05-05:
Tried latest (832) and same - it suddenly blanks out. I have only one Screen, no switching, not battery Attached, PAL.

thecodingfrog, 2016-05-06:
OK, I have tested with 4 screens and it works, so bug is happening when number of screens is lower

eiswuerfel, 2016-05-08:
Same for me. How did you set 4 screens? I uploaded all 4 tabs, but still the same.

thecodingfrog, 2016-05-09:
You have to set to 4 screen in config tab

tschoffy, 2016-05-10:
Same issue here though set to 4 screens in the config tab. Not happen allways when disarming but sometimes.
Any other idea someone to get rid of this Problem?

thecodingfrog, 2016-05-13:
Works fine using 824

night-ghost, 2016-05-17:
fixed in v834

closed 2016-05-17 by night-ghost