night-ghost/minimosd-extra #136: Please resize heading band to allow centering above artificial horizon

aheyer, 2016-05-19:
The artificial horizon has a proper width to be placed centred on screen. The heading band is one character to small/width, so you can't place it centred and aligned to the attitude indicator.

thecodingfrog, 2016-05-24:
It has to be like this because of centering needs. If you increase it to 12 chars how will you center North ?

thecodingfrog, 2016-05-24:
Ok I've done it, it's not so bad. I changed few things by the way and updated charset to make baseline more confortable

aheyer, 2016-05-25:
After posting my enhancement suggestion I thought the same: How to center a direction character with an even size? But it's possible like mwosd project shows. Seems you have to split characters into two positions.

Unfortunately your recent change did not correct the issue. You still can't line up the center of the artificial horizon to the center of the heading rose. Look at the GUI screen-shot:

[image]

The new arrow can't be centred. Personally I even don't feel a need for this arrow because I'd like to put the heading degree number above the rose and have the decimal ten position at center.

There's also another bug show in the screen shot: Look at the right upper edge. Under the time I have the trip's travel distance. Both fields have the same width in the GUI but if I place them aligned, the unit mark shifts one position to the right. So I have to displace the travel field to get right alignment.

thecodingfrog, 2016-05-25:
Maybe I can fork this project to share my mods, but I'am a newbie git user

thecodingfrog, 2016-05-25:
I forked the project if you want to try it out

thecodingfrog, 2016-05-28:
[image]

aheyer, 2016-05-28:
Looks good. Only problem now: You need 3 lines instead of 2. Personally I don't need the marker, it just suffice that the band is centred now. If I use the marker it has be be mirrored to be put under the rose band. Because I place the heading on top of the screen.mirrored to your style. So we need an additional mirrored center mark. :)

Thank you for your efforts. The best way to spread your work is to send a pull request to night-ghost. I'm in doubt that many users will change to your fork.

Do you have any clue what the new symbol in front of the heading is? This double arrow in my screen or the unidentifiable symbol in yours?

thecodingfrog, 2016-05-28:
the center mark can be removed, just click the option for heading rose. I'm afraid we can't do a bottom center mark, as we have to use chars from charset file that is limited.
In front of the 18° I have the arrow, it appears blured on this photo

night-ghost, 2016-05-31:
Guys, the screen has an even number of columns, and you'll never  center odd-chars panel.

Low marker I'll made

night-ghost, 2016-05-31:
see v846

closed 2016-05-31 by night-ghost