night-ghost/minimosd-extra #137: No input data v. 839

DrBraXo, 2016-05-22:
На версиях 837, 838 minimosd работает в штатном режиме, при обновлении до версии 839 данные телеметрии перестают отображаться.
При старте minimosd сначала на пару секунд появляется нужная страница, затем надпись no input data.

Pixhawk, arducopter 3.3.3.

Если нужна какая-то инфа для дебага, сообщу дополнительно.

tops4u, 2016-05-22:
According to Google Translate this means that OSD Data is not read correctly after a few secs with Version 839. I had the same Problem, why I "downgraded" to 836 which worked for me with the very same setup. 839 just showed "No Data". 

AUAV-X2, PX4 Daily Build from 21.5.16

miferpa, 2016-05-22:
Same is happening to me, I downgrade to v383 and works fine. On test bench I get a funny sign on vertical speed, I'l test it in flight to see what happen.

night-ghost, 2016-05-31:
fixed

closed 2016-05-31 by night-ghost