night-ghost/minimosd-extra #141: Multiwii Build?

alexanderad, 2016-06-04:
I was wondering if this fork is compatible with multiwii? 

I tried to flash one of latest builds to `minim-osd v1.1`, got working overlay and video while connected to PC. When I plug it in into multiwii, there is no overlay, just video.

Does multiwii requires some sort of a special build?

alexanderad, 2016-06-04:
I see that one of build targets is actually multiwii:
```sh
make -C $SRC PROTO='-DUSE_MWII=1'
[ -f $BUILD/MinimOsd_Extra.hex ] && mv $BUILD/MinimOsd_Extra.hex $RELEASE/MinimOsd_Extra_Uni.${VERS}DV-MWII-release.hex
rm -rf $BUILD
```

But I failed to build it:
```
make: Entering directory '/home/alex/multicopters/minimosd-extra/MinimOsd_Extra'
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [AUTODETECTED]       ARDUINO_DIR = /usr/share/arduino 
- [COMPUTED]           ARDMK_DIR = /usr/share/arduino (relative to Common.mk)
- [AUTODETECTED]       ARDUINO_VERSION = 105 
- [DEFAULT]            ARCHITECTURE =  
- [DEFAULT]            VENDOR = arduino 
- [DEFAULT]            ARDUINO_SKETCHBOOK =  
- [BUNDLED]            AVR_TOOLS_DIR = /usr/share/arduino/hardware/tools/avr (in Arduino distribution)
- [COMPUTED]           ARDUINO_LIB_PATH = /usr/share/arduino/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_VAR_PATH = /usr/share/arduino/hardware/arduino//variants (from ARDUINO_DIR)
- [COMPUTED]           BOARDS_TXT = /usr/share/arduino/hardware/arduino//boards.txt (from ARDUINO_DIR)
- [USER]               USER_LIB_PATH = ../libraries 
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_TAG = atmega328 
- [COMPUTED]           CORE = arduino (from build.core)
- [USER]               VARIANT = standard 
- [COMPUTED]           OBJDIR = build-atmega328 (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = /usr/share/arduino/hardware/arduino//cores/arduino (from ARDUINO_DIR, BOARD_TAG and boards.txt)
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [USER]               CFLAGS_STD = -std=gnu11  -fno-exceptions  -Wl,--gc-sections -fno-threadsafe-statics -ffast-math  -floop-strip-mine -finline-functions-called-once  -finline-small-functions -fmerge-all-constants -fomit-frame-pointer  -fno-optimize-sibling-calls  -fno-early-inlining  -fno-caller-saves -DUSE_MWII=1 -Wl,--relax  
- [USER]               CXXFLAGS_STD = -Wno-deprecated-declarations -std=gnu++11  -fno-exceptions -Wl,--gc-sections -fno-threadsafe-statics -ffast-math  -floop-strip-mine -finline-functions-called-once  -finline-small-functions -fmerge-all-constants -fomit-frame-pointer  -fno-optimize-sibling-calls  -fno-early-inlining  -fno-caller-saves -DUSE_MWII=1 -Wl,--relax 
- [AUTODETECTED]       DEVICE_PATH =  
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
-
-                      ARDUINO_LIBS =
- [USER]                 GCS_MAVLink
- [USER]                 SingleSerial
- [USER]                 TimerSerial
- [COMPUTED]           BOOTLOADER_PARENT = /usr/share/arduino/hardware/arduino//bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.5 
- [COMPUTED]           CC_VERSION = 4.9.2 (avr-gcc)
-------------------------
mkdir -p build-atmega328
/usr/share/arduino/hardware/tools/avr/bin/avr-g++ -MMD -c -mmcu=atmega328p -DF_CPU=16000000 -DARDUINO=105  -D__PROG_TYPES_COMPAT__ -I/usr/share/arduino/hardware/arduino//cores/arduino -I/usr/share/arduino/hardware/arduino//variants/standard     -I../libraries/GCS_MAVLink   -I../libraries/SingleSerial   -I../libraries/TimerSerial -Wall -ffunction-sections -fdata-sections -Os -fno-exceptions -Wno-deprecated-declarations -std=gnu++11  -fno-exceptions -Wl,--gc-sections -fno-threadsafe-statics -ffast-math  -floop-strip-mine -finline-functions-called-once  -finline-small-functions -fmerge-all-constants -fomit-frame-pointer  -fno-optimize-sibling-calls  -fno-early-inlining  -fno-caller-saves -DUSE_MWII=1 -Wl,--relax ArduCam_Max7456.cpp -o build-atmega328/ArduCam_Max7456.o
ArduCam_Max7456.cpp:1:0: sorry, unimplemented: Graphite loop optimizations cannot be used (-fgraphite, -fgraphite-identity, -floop-block, -floop-interchange, -floop-strip-mine, -floop-parallelize-all, and -ftree-loop-linear)
 #include "compat.h"
 ^
/usr/share/arduino/Arduino.mk:1197: recipe for target 'build-atmega328/ArduCam_Max7456.o' failed
make: *** [build-atmega328/ArduCam_Max7456.o] Error 1
make: Leaving directory '/home/alex/multicopters/minimosd-extra/MinimOsd_Extra'

```

nathantsoi, 2016-06-04:
Scarab osd is more up to date :)

alexanderad, 2016-06-04:
Scarab works for me, but I like this one for the GUI.  Nice panel layout configuration and better fonts.

alexanderad, 2016-06-04:
I tried to comment out unsupported optimization params but build still fails with bunch of compile errors this time. This leads to my original question -- is multiwii initially supported by this fork or it is built solely for mavlink?

Johnex, 2016-06-04:
Multiwii has been added yes, but it is untested. Night_Ghost only has APM + Pixhawk and Devo for testing. Write him in the main forum thread about your issues.

night-ghost, 2016-06-04:
to build project via Arduino.mk you should use gcc 5.3

night-ghost, 2016-06-21:
now I have OpenPilot Atom and want to flash Cleanflight to it to debug MWII protocol. Can anyone give me instructions about this?

Johnex, 2016-06-28:
Night-ghost, you should download Google Chrome, and install Cleanflight from here: https://chrome.google.com/webstore/detail/cleanflight-configurator/enacoimjcgeinfnnnpajinjgmkahmfgb?hl=ru-Ru

night-ghost, 2016-06-28:
Cleanflight already flashed and visible by native CT.  But I can't get telemetry data flow from it.

Johnex, 2016-06-28:
I don't know exactly which telemetry you need, but to enable frsky you should run some CLI commands:

feature SOFTSERIAL
feature TELEMETRY
set softserial_baudrate=9600
set softserial_1_inverted=1
set telemetry_port=1
set telemetry_provider=0

Johnex, 2016-06-28:
If its something else here is the telemetry docs for cleanflight:

https://github.com/cleanflight/cleanflight/blob/master/docs/Telemetry.md

night-ghost, 2016-06-28:
Thanks but I already read it from left to right and right to left :) I need native MWII telemetry which used by MWOSD/SCARAB-OSD https://github.com/ShikOfTheRa/scarab-osd/blob/master/OTHER/DOCUMENTATION/User_Guide.md

night-ghost, 2016-06-28:
By the way, I successfully got LibrePilot telemetry - but only when USB connected. somehow come to testing :)

Johnex, 2016-06-28:
https://drive.google.com/file/d/0B2MInRUrbpWxUjdmdXJIWkRZY0k/view

night-ghost, 2016-06-28:
I do not understand why you give me a link to the document that I just have that?

x2, 2016-07-22:
I think the problem is that CleanFlight's firmwares are using **passive** telemetry. That means OSD have to continuously send FC commands for it to output telemetry data. MWOSD works this way.

x2, 2016-07-23:
Yep, I've got the data flow from my NAZE32 (basically a MultiWii on STM board) to the OSD by connecting a separate arduino that was sending MSP commands to the FC. I've also had to lower the multipliers for attitude values and change X and Y axes. I'll try to check other values when I get the weather to fly.

night-ghost, 2016-09-05:
latest release can request datastream from MWEE-based controllers

x2, 2016-09-06:
Wow! Thanks a lot! It is now working from the box! Roll&pitch axes are correct now. But those still look to have too much gain on my board. I have to multiply its values by 0.1 to get correct attitude on my CleanFlight's NAZE32. 
 I'd also suggest to increase frequency of attitude requests (like 5x of current value or even greater) lowering other values request rates if needed.

night-ghost, 2016-09-27:
I don't use this FC so I need a bit more details like "there should be XXX but OSD shows YYY", scales can be wrong because I can't find complete documentation on MWII protocol.

rate request logic comes from mwosd project "as is" and can't be adjusted. I plan to rewrite this part in near future.

night-ghost, 2016-11-12:
seems to me that question is over

closed 2016-11-12 by night-ghost