night-ghost/minimosd-extra #145: Smaller font?

MacPiston, 2016-06-11:
When original minimOSD firmware was on top, there were .mcm files with smaller fonts. Today I was searching, and found one that almost works properly with nightghost FW. Only thing I saw is that artificial horizon is bugged - idk if something else, because im not using it. Please look at the file (change extension to .mcm instead of zip)
[MinimOSD_2.4.0.2.zip](https://github.com/night-ghost/minimosd-extra/files/310375/MinimOSD_2.4.0.2.zip)
 and think about it. Cheers!

night-ghost, 2016-06-21:
This is good but current MCM wizard don't allows to copy characters from one font to another so it needs to redraw a lot of chars. Don't forget to change font version in PR

MacPiston, 2016-06-21:
Yeah I have already redrawn them, but they don't want to be uploaded on
minim... I'll find them on PC and send redrawn file to you.

night-ghost, 2017-12-21:
more than year, so close

closed 2017-12-21 by night-ghost