night-ghost/minimosd-extra #148: Timer starts from 4:40, not 0:00 like it was before

ins2d, 2016-07-09:
As simple as that. It's was (0:00) at v850DV (downgrade helps) but starts from some crazy numbers @ v858, v860, v862 (shows 861 in CT BTW).

Screenshot:
http://prntscr.com/bqlmvo

Video:
https://www.youtube.com/watch?v=QH4aZ306djk

My setup is:
MinimOSD micro; APM @ AC 3.0.1

Also mentioned by another user at RCG:
http://www.rcgroups.com/forums/showpost.php?p=35204930&postcount=1144

TNX!

tops4u, 2016-07-09:
Exactly same Problem occured here with MicrominimOSD, AUAV-X2, PX4 1.4.0 RC2. Using 850 now where this works - @ins2d thanks for the hint.

night-ghost, 2016-07-11:
There is strange bug somewhere in GCC which cause corruption of some memory region. When most of optimisation options are turned off this bug gone but all functions don't fit to flash.

In 863-864 time is OK, corrupted region changed to MAVlink data which quickly overwrites bad data.

ins2d, 2016-07-11:
Hey man!

I've just tested v864-mavlink and indeed Time panel works now as expected. At least for me!

Here is my screenshot (Timer is at the very bottom):

http://prntscr.com/brpc98

Setup: Micro MinimOSD, APM @ AC 3.0.1, HS1177 PAL (GoPro w/ NTSC was tested too). Only bench testing, no actual flight.

Thank you, that was fast!!!

closed 2016-07-12 by night-ghost