night-ghost/minimosd-extra #155: Home distance not correct

0n3-70uch, 2016-09-05:
Hello,

I found a little bug in Mavlink Copter Version 881. When I start my drone, fly a round and Land then the distance is shown correct. After that I picked up my drone without removing the battery and Start at another Position. Now the osd Show me the Position to the First startpoint (for example 1500m).

That's all... Everythink Else works great Thank you.
Greetings
Fabi

night-ghost, 2016-09-15:
this was a feature - home location fixed on 1st arming only to exclude a possibility to get wrong home point in flight. 

But... I can reset home lock on disarm :)

0n3-70uch, 2016-09-15:
Haha ok. "This is no bug this is a feauture" ;).

But when I disarm and rearm then the distance is also shown incorrect...

night-ghost, 2016-09-28:
You should disconnect battery when move to a new location.

closed 2016-09-28 by night-ghost

0n3-70uch, 2016-09-28:
Thanks for the new option.