night-ghost/minimosd-extra #161: Request for Deg mark to home

Ansarida, 2016-10-12:
Hi Night Ghost

The Home direction point must have degree number right?
can it been show the number degree? 

It can be used to aim manually antenna

night-ghost, 2016-10-18:
I do not understand, what is the relationship between the direction to a house in the copter's frame and the direction of the copter in the operator's frame. True tracker is still does a search by the RSSI level

Ansarida, 2016-10-20:
Hi Night Ghost
Thank for reply.

is manual tracker.. change by hand of the pilot, no matter the quad/plane yaw it mush pointing at home , At operator side only need look at degre number on osd then toward the antena using hand at the degree number showing at osd..

It will easy to know the quad is right or left

night-ghost, 2016-10-20:
digital form will be in next release

closed 2016-10-20 by night-ghost

Ansarida, 2016-10-21:
ok sir thank you very much... u relly cool man