night-ghost/minimosd-extra #162: ""No input data" message on the monitor from minimosd-extra board running under FW V889

sergeykorn, 2016-10-20:
Though it's not a bug report..... )

Minimosd board connected to Telemetry2 port on PIXHAWK Flight controller gives "No input data" message. After pushing the reset button on the board, the screen updates and shows relevant content for a second or so and then goes back to the "No input data" message.

[image]

sergeykorn, 2016-10-28:
I've just found that if I transmit data through COM port in Config_OSD tool the screen goes as it supposed to be. As soon as the transmission is stopped No Input data message pops up again!

night-ghost, 2016-11-10:
>As soon as the transmission is stopped No Input data message pops up again!

you must be joking! YES if transmission is stopped OSD REALLY hasn't any input data

closed 2016-11-10 by night-ghost

sergeykorn, 2016-11-12:
I meant a transmission goes through while OSD board is connected to PC via USB to TTL adapter. It's giving the same No Input Data msg when connected to Pixhawk telemetry port anyway! Why it's not showing the screen set up in configuration tool with the chosen graphics and zero values for variable parameters if actually there is no input data?!

mP1, 2016-11-13:
@sergeykorn 

Im having basically the same problem, and found a solution. I no my setup was correct as sometimes the "no mavlink data" screen would show my gps coords. After i did change my params as shown below the osd just worked ;)

Issue 121 (here) has the solution.

> you have to disable either BRD_SER1_RTSCTS or BRD_SER2_RTSCTS by setting it to 
0 (or maybe both). 

https://github.com/night-ghost/minimosd-extra/issues/121

Hope that helps...

sergeykorn, 2016-11-16:
Thanks, I will try these settings.

sergeykorn, 2016-11-16:
Doesn't help :(

Zethyl, 2017-09-14:
worked for me on Plane 3.8.2 with minimosd-extra MAVLINK934DV