night-ghost/minimosd-extra #164: Adjust pin value not as accurate as needed

Gesiima, 2016-11-03:
Hi, I am trying to attach a APM Power Module to my minimOSD like this one.
http://www.banggood.com/APM-Power-Module-V1-Vision-50V-90A-with-BEC-for-RC-Model-p-917309.html
I get values for current and voltage but I am not able to calibrate my messures.
For voltage I need a adjust pin value like 10.4 but a dot is not possible.
The value for current is about 18.2Amps per Volt. What is the best way to find out the adjust pin value for that?

night-ghost, 2016-11-10:
proof?

night-ghost, 2016-11-13:
>For voltage I need a adjust pin value like 10.4 but a dot is not possible.

really?
[image]

closed 2016-11-13 by night-ghost

Gesiima, 2016-11-16:
Only triyed dot . not komma , 
Shame on me!
Thanks alot.

night-ghost, 2016-11-16:
dot or comma is dependent on your locale settings and NET/MONO layer