night-ghost/minimosd-extra #165: Blank screen 3 shows warning label

athertop, 2016-11-06:
f/w version: MinimOsd_Extra_Uni.893DV-MAVLINK-release

I have 3 screens configured - screen 3 is configured to be blank (no data items selected from list), but yet still shows the warnings item top right.
Also my 2nd screen shows the flight data item - 3rd item in this list displays additional data way off to the left.

night-ghost, 2016-11-10:
yes warnings shown at all screens by design

closed 2016-11-10 by night-ghost

athertop, 2016-11-12:
NG - I don't think you are giving my comments enough consideration, and I maybe could have provided more specific info in my report so here goes: On the 3rd screen with nothing configured, the warning displays in the top right corner of the screen half-in frame and half out of frame, and not in any useful position at all. It looks like an oversight in the code. This could be better presented IMHO!
Second point - in which way can we create a blank screen (simulating the OSD is switched off), if we are forced to display data we don't want to see on a screen containing no assigned overlays?

Please, before responding, can I suggest you have a look at the issue. A lot of your responses to reported issues in here appear to be less than helpful and in some cases quite disrespectful. Please understand that by reporting issues we are trying to help in the further improvement of the project - make it better for everyone. Having a really well presented OSD is of great importance to me. 
I think for anyone, seeing a project with a keen dev(s) who actively welcome feedback from testers, is absolutely essential to maintain those peoples' interest in the project. By continually bashing reports from users, you are undermining this interest and good will, and ultimately causing the project to get a bad reputation (do you remember baseflight?). This is the reason why I started the move away from MinimOSD in the first place - and found other interesting OSD projects like AlceOSD and PlayUAVOSD. 

Please don't misunderstand me, the work you have done here is pretty incredible, and I respect you for sticking with the project so long, but please consider that those who report issues are initially trying to help further the project - so giving responses more thoughtful consideration would really help keep their interest in the project, and the project popular.

Best wishes, Paul A (athertop)

night-ghost, 2016-11-12:
Great people discuss ideas. Ordinary people discuss events. Stupid people are discussing people. (C)

You do not bother to properly draw a bug report and began to discuss me. I will not bother you.

athertop, 2016-11-12:
I would have expected nothing less from you in response! 
Here's an idea - be a nicer person! It will help your project a lot!

nathantsoi, 2016-11-12:
https://github.com/ShikOfTheRa/scarab-osd is under active development by lots of nice people @athertop

athertop, 2016-11-12:
Cheers for the link Nathan. I am running ardupilot unfortunately, but the OSD looks good. I am slowly moving all my models onto PlayUAVOSD. Just one to go!