night-ghost/minimosd-extra #166: mavlink generate

leomatrix, 2016-11-11:
Hi:
         Excuse me. I want to add some mavlink message, and I modify the file of ./libraries/GCS_MAVLink/generate.sh and remove exit at the start.
         but I find out that The mavlink v1.0 generated by me is not match your code.  which version of mavgen.py do you used ????
         could you give me you email address! I want to consult you some question ! Thank you very much!

thecodingfrog, 2016-11-11:
If you need to add message to mavlink you have to add it to common.xml file so it will be automaticaly generated using mavgen. But mavlink 1.0 is limited to 255 message definition ans bas already reached it so you'll have to replace an unused message. Then you'll have to modify your own version of arducopter. I did it but I failed at it. My arduino didn't get the message ID as expected. I'm waiting some hardware to be able to output debug info

night-ghost, 2016-11-12:
MAVLink here is not generated but altered by hands, including additions of new messages

night-ghost, 2016-11-12:
and this is definetely not an issue of this firmware

closed 2016-11-12 by night-ghost

leomatrix, 2016-11-15:
I find the the reason is the common.xml and ardupilotmega.xml is doesn't match the mavlink v1.0 at your raw code, so if I use a laster version of common.xml and ardupilotmega.xml the problem could be resolved.

night-ghost, 2016-11-15:
generated MAVlink eats out ~400 bytes of RAM and a lot of flash so I don't need to "resoive a (contrived) problem". Now this library is highly optimized, and to add a new message by hands takes a couple of minutes.