night-ghost/minimosd-extra #167: the generate hex file is too large to update firmware

leomatrix, 2016-11-15:
#### Issue details
_Please describe the problem, or desired feature_
           My problem is that I git clone your newest version code of minimosd-extra, and use ./build.sh to build the raw code, firstly I found there was a error, then I modify the file of Config.h to uncomment the line of #define USE_MAVLINK 1 , and then code can be build successfully!
          After building (I don't modify anything of this project)I was read to flash the new code to  minimosd, I use the OSD_Config.exe in your project(the version is MinimOSD-Extra Config Tool 2.4.1.5 r894 DV), But this tool told me that "file is, too big", the MinimOsd_Extra_Uni.893DV-MAVLINK-release.hex generate by my crosstoll(arm-none-eabi-gcc) is 89.4kB， and the hex at you project is 85.8kB only.
          I also attempt use another version of OSD_Config.exe to flash MinimOsd_Extra_Uni.893DV-MAVLINK-release.hex generated by me, I find that after update firmware the minimosd can't work normally...
         I have use the arduino 1.0.6 IDE to build  the code of https://github.com/diydrones/MinimOSD-Extra.git and update firmware for minimosd, I find this firmware could work normally.
         could you give me some tips. Thank you very much!

After make, the shell told me :
**AVR Memory Usage
----------------
Device: atmega328p

Program:   31764 bytes (96.9% Full)
(.text + .data + .bootloader)

Data:       1585 bytes (77.4% Full)
(.data + .bss + .noinit)**

#### Version
_What version was the issue encountered with_   night-ghost/minimosd-extra/ 's master branch

#### Board type
_What type of the OSD board you are using?_  minimosd

#### How "canonical" OSD firmware works?
[  ] Works fine
[  ] The same bug
[  ] Not applicable
[ * ] Not tested

#### TLOG and OSD
_Please provide a links to a relevant screenshot/video, TLOG and OSD files that show the issue_

night-ghost, 2016-11-16:
this is issue of your build system, not firmware, Closed.

PS, comment out #define's for unneeded features

closed 2016-11-16 by night-ghost

leomatrix, 2016-11-16:
Could you tell me which build system do you used???

night-ghost, 2016-11-16:
* patched Arduino libs
* Makefile instead of Arduino IDE (see *.sh files)
* use proper GCC version

leomatrix, 2016-11-16:
could you give me a tip how can I get the patched Arduino libs ??
which GCC version do you used??

night-ghost, 2016-11-16:
1. see my issues to Arduino project, also comment out almost all work with timers in wiring*.c
3. gcc 4.9.  gcc 5.4 gives little smaller code but with a lots of errors. Building avr-gcc from source documented somewhere in Arduino's github

AND AGAIN, just comment out unneeded features!

leomatrix, 2016-11-16:
1. I think patched Arduino libs you mensioned is  the "libraries" folder in the root path of your project, Am right or not??
2.I have used the arm-none-eabi-gcc 4.84 and 4.93 to build the code. but It seems not work well...

night-ghost, 2016-11-16:
1. no, a lot of files in hardware/arduino/avr/cores/arduino
2. # avr-gcc -v
Using built-in specs.
Reading specs from /usr/share/arduino-1.6.12/hardware/tools/avr.orig/bin/../lib/gcc/avr/4.9.2/device-specs/specs-avr2
COLLECT_GCC=/bin/avr-gcc
COLLECT_LTO_WRAPPER=/usr/share/arduino-1.6.12/hardware/tools/avr.orig/bin/../libexec/gcc/avr/4.9.2/lto-wrapper
Target: avr
Thread model: single
gcc version 4.9.2 (GCC)

leomatrix, 2016-11-17:
I used "sudo apt-get install arduino-mk" , but I find the version of avr-gcc is 4.8.2 by apt-get install method？My linux-pc version ubuntu014.04LTS, how should I install avr-gcc 4.9.2

leomatrix, 2016-11-17:
I use "sudo apt-get install arduino-mk" to install the the cross-tools, I see actually these pachkages(gcc-avr binutils-avr gdb-avr avr-libc avrdude) were installed.
But the avr-gcc version is 4.8.2 using the apt-get install method, then I modify the  /usr/share/arduino/hardware/tools/avr/bin(this is a symbolic link) to point to a 4.9.2 version`s bin path. and I check the version of /usr/share/arduino/hardware/tools/avr/bin/avr-gcc , I can make sure that I have used 4.9.2 version's avr-gcc avr-g++ and avr-objcopy to build the code. But the hex file I buid is still 88kB, It can't   be flashed to minimosd.  Could you give me some help ???

night-ghost, 2016-11-17:
You must be kidding! 

JUST COMMENT OUT UNNEEDED FEATURES!!!!

leomatrix, 2016-11-18:
Now I use 4.9.2 building system, I find that if I use USE_MWII only(uncomment #define USE_MWII 1 line only), I can generate a hex file only 79kB, this could be flash to minimosd hardware , and I can see the yellow led blink all the time. But if I use USE_MAVLINK only, the hex I generate will be 88kB, this one can't be flash to miniosd hardware, if I use another tools to flash, the minimosd will dead(yellow led not blink),

you say "JUST COMMENT OUT UNNEEDED FEATURES!!!!"

I see the MinimOsd/Config.h file, there seems only very much feature I can comment... I have comment "USE_SENSORS" , Could you tell me are there other option can be comment!!!

Thank you very much!!!

leomatrix, 2016-11-18:
could you give me a copy of output file when you run ./build.sh ! My email is : guweiyangpeng@126.com