night-ghost/minimosd-extra #169: HS1177 Cam causes OSD to crash

garthberry, 2016-11-25:
#### Issue details
When micro minimosd (MAVLINK) is attached to a HS1177 (mini) the OSD will crash if any input is set to "from FC". The overlay is seen on power up for less than a second and the data transmit LED stops blinking. If the camera is unplugged before powering on the overlay data can be seen updating, plugging in the camera "hot" will also cause a crash. I have also tried NTSC, PAL and AUTO video modes with the same result. Swapping the camera out for a Aomway 700TVL WDR enables the OSD to function without problems.

#### Version
880DV-MAVLINK
884DV-MAVLINK
901DV-MAVLINK
904DV-MAVLINK
905DV-MAVLINK

#### Board type
Micro MinimOSD

#### How "canonical" OSD firmware works?
[  ] Works fine
[x ] The same bug
[  ] Not applicable
[  ] Not tested

#### TLOG and OSD
The issue can be reproduced using default configuration and the TLOG file included with the osd_latest.zip package. At least one input (V1, V2, Current) must be set to "From FC".

I will update the issue shortly with video.

Shotfire, 2016-12-03:
garthberry: Thanks for brining up this issue. I've pulling my hair out for two weeks on and off trying to get it to work with a 1177 and two different variant's. Same problem: Boot screen appears, then no data rx light or OSD but video feed is fine. I've tried two different minimosd boards and same condition

garthberry, 2016-12-03:
@Shotfire No problem. I have been able to run the HS1177 with scarab-osd without any problems, so I am pretty sure hardware is ok. Unfortunately mavlink support in the scarab-osd doesn't seem to be great.

Shotfire, 2016-12-03:
@garthberry  That's funny, I was just now checking that out, scarab-osd and saw that at least right now it only supports mavlink V1.

night-ghost, 2016-12-05:
I don't have HS1177 so can't test it

palladin82, 2016-12-06:
w/o cam the same issue

garthberry, 2016-12-06:
@night-ghost I have a spare HS1177, I can send it to you for testing?

palladin82, 2016-12-07:
h/w like this https://yadi.sk/d/Kv3-xIG_32SQUT
tried any fw before 880, the same( data transmit LED stops blinking)
загрузчик нормально рисует версию, после загрузки происходят моргания экрана монитора и отсутствие миганий светодиода

night-ghost, 2016-12-07:
> w/o cam the same issue
this means is a some sort of hardware problem. Check supply voltage, try to change OSD board itself.

>I can send it to you for testing?

Do you have an oscilloscope? Camera's VSync is used as interrupt so if MAX receives wrong video it may do noise on interrupt line and in result hang program.

>  I have been able to run the HS1177 with scarab-osd without any problems

ScarabOSD has a lots of config options and I don't know which was used in your test.

palladin82, 2016-12-07:
> this means is a some sort of hardware problem. Check supply voltage, try to change OSD board itself.
other ver. of extra working fine (http://diydrones.com/forum/topics/adding-extra-functions-to-minimosd)
попробую скачать весь проект, дебаги повставляю для медленной инициализации знакогенератора...
попытаюсь сам разобраться, потом расскажу, аж самому интересно

night-ghost, 2016-12-07:
>other ver. of extra working fine

That version don't uses interrupts at all and simply writes directly to MAX

for russian please come [here](http://www.ykoctpa.ru/groups/eye-in-a-sky/forum/topic/minimosd-novaya-proshivka/?topic_page=28&num=15#post-9489)

garthberry, 2016-12-07:
@night-ghost I do have an oscilloscope, I can take a look. 

The OSD only crashes when any input is set to "from fc" and data is received from FC. The example I showed in the video is repeatable and same across 2 micro minimosd boards.

scarab-osd config options, I have attached the config.h. I am currently using this firmware with 2 other micro minimosd/HS1177 cam combinations. They are both attached to boards running betaflight though. In the configurator all inputs are set to "from fc", none of the micro minimosd board sensor inputs are used.

[Config.h.zip](https://github.com/night-ghost/minimosd-extra/files/636288/Config.h.zip)

night-ghost, 2016-12-07:
> I do have an oscilloscope, I can take a look.

Excellent! What is on pin 32 of ATMega with different cameras? This in PD2, Int0

> The OSD only crashes when any input is set to "from fc" and data is received from FC

this is a very, very strange! When this flags in "From FC" state then there is MUCH LESS work in loop().

night-ghost, 2016-12-07:
Just make a test version 907, try it with problematic camera

palladin82, 2016-12-07:
For me test going good, not blinking

night-ghost, 2016-12-08:
When there is no interrupts then OSD automatically switches to screen update on each 20ms but in this case there will be a "snow" on the screen. Moreover, without interrupts changes that were made in 907 should not help.

Support for boards without VSYNC will be in the next version

SlipstreamFPV, 2016-12-19:
HS1177 hasn't worked with MinimOSD in the past as well. 

As other pointed out it doesn't appear to be a hardware issue as MWOSD / ScarabOSD etc all work on the same hardware with the camera just fine.

I can get mine to work if I let the OSD boot first and then power the camera second.

night-ghost, 2016-12-26:
no oscilloscope data that I asked, no necessary information. sorry :)

closed 2016-12-26 by night-ghost