night-ghost/minimosd-extra #171: no input data

zacool, 2016-12-18:
#### Issue details
_Please describe the problem, or desired feature_

#### Version
_What version was the issue encountered with_

#### Board type
_What type of the OSD board you are using?_

#### How "canonical" OSD firmware works?
[  ] Works fine
[  ] The same bug
[  ] Not applicable
[  ] Not tested

#### TLOG and OSD
_Please provide a links to a relevant screenshot/video, TLOG and OSD files that show the issue_

zacool, 2016-12-18:
i have a crius minimosd v2.2 on an apm 2.8 fc and i update to the last v9.12 but it stays on a screen like fly data screen and it says NO INPUT DATA......every 65 sec turns to 1 screen for 2 sec but without data.....

night-ghost, 2016-12-26:
there is no necessary information

closed 2016-12-26 by night-ghost