night-ghost/minimosd-extra #174: Strange battery warning behaviours

pulquero, 2017-02-03:
#### Issue details
1) Battey capacity flashes 0 for full battery when min voltage warning is set to 0
2) For fully charged 5000mah battery, low voltage warning occurs with min voltage warning set to 11 and remaining % warning set to 25. On screen shows 12.8v and 100%.
3) For used 3000mah battery, low voltage warning is not shown with min voltage warning set to 11 and remaining % warning set to 25. On screen shows 11.3v and 18%.

#### Version
2.4.15 r916, 918

#### Board type
Micro MinimOSD

#### How "canonical" OSD firmware works?
[  ] Works fine
[  ] The same bug
[X] Not applicable
[  ] Not tested

night-ghost, 2017-02-11:
where video, tlog and OSD files?

pulquero, 2017-02-11:
I'm using batt A pin, not FC (no battery telemetry in mavlink data).

night-ghost, 2017-02-13:
Ok. where video and OSD files?

night-ghost, 2017-03-21:
no needed files in 2 weeks so closed

closed 2017-03-21 by night-ghost