night-ghost/minimosd-extra #175: xjt transmitter s-port not compatible.

triolus, 2017-02-11:
Composite video to goggles/monitor being hacked into OSD. This is to remove the osd from the craft. No option to take xjt transmitter s-port telemetry to osd. Osd will not overlay over video on test. when video receiver is off the osd data is available, but missing cheveron.  Requesting direct frsky s-port to OSD option without flight processor in between.

night-ghost, 2017-02-13:
you must be joking! But PRs are welcome

closed 2017-02-13 by night-ghost

triolus, 2017-02-13:
Nope. wasn't joking. xjt does transmit s-port signal out the back for an outboard monitor for telemetry. If this is the case, a craft and theory cable can be used to change uart to s-port signal. Hooked it up. One problem is that when the xjt is on, it gives a black screen with data. but when off, I can see the camera coverage. I think this can be repurposed from another github. It isnt finished, but close. Most of it is done and is here: https://github.com/sixtemesseven/TDOG/tree/v0.9 and here: http://diydrones.com/profiles/blogs/minimosd-based-groundstation.

triolus, 2017-02-13:
This person may have it completed, or is close: https://github.com/razvanstanga/minimosd-frsky

night-ghost, 2017-02-13:
It seems you are messed up free software with free execution of orders.

Project by links above - are a perversions and misuse of equipment, I'm not interested.