night-ghost/minimosd-extra #176: On pixhawk 2 params error on connecting tx pin

triolus, 2017-02-17:
On boot-up get error of Params 12 in Ardupilot. Upon disconnecting OSD the issue goes away. Others have pulled the TX going from the Flight controller, but control of the PWM switch might not work. 
_Please describe the problem, or desired feature_

#### Version
_What version was the issue encountered with_
918 mavlink

#### Board type
_What type of the OSD board you are using?_
Pixhawk Micro OSD from banggood

#### How "canonical" OSD firmware works?
[  ] Works fine
[  ] The same bug
[  ] Not applicable
[  ] Not tested

#### TLOG and OSD
_Please provide a links to a relevant screenshot/video, TLOG and OSD files that show the issue_

night-ghost, 2017-03-04:
where proof files?

night-ghost, 2017-03-21:
no needed files in 2 weeks so closed

closed 2017-03-21 by night-ghost