night-ghost/minimosd-extra #178: Request: New panel to indicate when camera is triggering

iskess, 2017-02-26:
#### Issue details
I would like to be able to see when my mapping camera is triggering. This has already been done here: https://github.com/ArduPilot/ardupilot/pull/1626
Most of us don't have camera feedback setup, but would simply like an indication that the pixhawk is sending out camera triggering signals. 

[image]
[image]

night-ghost, 2017-03-04:
It is unlikely that most users need this feature, but there is paid support for business opportunities

night-ghost, 2017-04-17:
closed - no answer

closed 2017-04-17 by night-ghost