night-ghost/minimosd-extra #179: time error with negative TZ offset

tmaclean, 2017-03-21:
#### Issue details
When I set a timezone offset that shows a time in the prior day, the displayed time is not correct.
E.g. at 2:24 GMT, with offset -4, I see 1090:40 displayed. A small offset like -2 (which is the same day, in this case) displays correctly.

#### Version
v920

#### Board type
CRIUS MinimOSD

#### How "canonical" OSD firmware works?
[  ] Works fine
[  ] The same bug
[  ] Not applicable
[x  ] Not tested

#### TLOG and OSD
_Please provide a links to a relevant screenshot/video, TLOG and OSD files that show the issue_

[image]

tmaclean, 2017-03-21:
Looks like the time in hours and minutes is stored as a UINT16.  1090 is (64K - 80) / 60 as UINT

night-ghost, 2017-03-21:
> time in hours and minutes is stored

Sure!

night-ghost, 2017-03-21:
fixed in v822

night-ghost, 2017-03-22:
v924
fixed size issue again. Flash is full :(

tmaclean, 2017-03-24:
Sad to say, MinimOsd_Extra_Uni.924DV-MAVLINK-release is still too big.    :-(

night-ghost, 2017-03-24:
just download it again

tmaclean, 2017-03-25:
Good news and bad.  The good is that the file size is better; the bad is that the issue is still there.  At 00:56 GMT, I read 1089:12 on the OSD using time offset -4.

night-ghost, 2017-03-25:
re-uploaded, plz check again

night-ghost, 2017-04-03:
silence more than week, closed

closed 2017-04-03 by night-ghost