night-ghost/minimosd-extra #180: When I try to Update Firmare with 9.23 Mavlink it keeps telling me file is to big

marvinlange, 2017-03-21:
#### Issue details
_Please describe the problem, or desired feature_

#### Version
_What version was the issue encountered with_

#### Board type
_What type of the OSD board you are using?_

#### How "canonical" OSD firmware works?
[  ] Works fine
[  ] The same bug
[  ] Not applicable
[  ] Not tested

#### TLOG and OSD
_Please provide a links to a relevant screenshot/video, TLOG and OSD files that show the issue_

night-ghost, 2017-03-22:
v924
fixed size issue again. Flash is full :(

marvinlange, 2017-03-22:
You are awesome I have been struggling with minimum OSD lack of features for years you have brought it back to life and made it incredibly use-full again !!!!!
GOD Bless You
Marvin

closed 2017-03-23 by night-ghost