night-ghost/minimosd-extra #182: Problem with Pixhawk "Get Params 12"

mastersurferde, 2017-03-25:
#### Issue details
Hi,
with actual night-ghost Version on MinimOSD I can´t connect to my pixhawks. The OSD has a 4 wire connection. The Mission Planner shows first "Get Params 1" an after collecting the parameters it shows "Get Params 12" an nothing happens until I stop the connection.
I have read from other people having the same problem. I testet on 2 Pixhawks (Quad and Hex with 3.4.6 Firmware) - always the same.
#### Version
I tried different versions. The last version working correct is the 894. Starting with 896 and later the problem happens.

Regards,
Stefan

night-ghost, 2017-03-26:
All works for me. Proof?

mastersurferde, 2017-03-26:
Hi,
some minutes ago I did a new test with complete different Equipment:
[image]
Pixhawk 2.4.8 with Arducopter 3.4.6 (my other Tests were made with Pixhawk lite - same result)
MinimOSD from Banggood with actual 924 Mavlink

Trying to connect to pixhawk:
[image]
It doesn´t matter if I try to connect with USB or with 3DR Radio on another serial port - always the same.
https://youtu.be/sIgmMs5Nz7A
MP tries to "Get params 12" until I cancel the action.

The guys here have the same problem:
http://diydrones.com/forum/topics/mission-planner-get-params-12
The first version not working correct, the 986 was released on 11.11.2016. So I think it is the same problem the guys on diydrones have.

I hope, you can help me.
Best regards,
Stefan

night-ghost, 2017-03-27:
This is very strange, OSD generates HeartBeat but uses IDs that comes from FC.
 Try v925

mastersurferde, 2017-03-27:
Hi, 
thank you very much, but with 925 I have no overlay. Tried back to 894 - this was ok.
Regards,
Stefan

night-ghost, 2017-03-27:
>no overlay

fixed in v926

mastersurferde, 2017-03-27:
Thank you,
actually it works :)
I will do some test in the next days.
Best regards.

closed 2017-03-27 by mastersurferde