night-ghost/minimosd-extra #188: r929 displays no data on screen

iskess, 2017-04-15:
#### Issue details
r929 displays no data. r928 and earlier work fine with all else being the same. 

#### Version
r929

#### Board type
generic minimOSD

night-ghost, 2017-04-16:
some bugs in GCC cause such  issues, I need to add some useless calls in Setup() to fix it. Somewhere I got this
[image]

somewhere this

[image]

just by adding one ''z" to line

`Serial.print_P(PSTR("#1zzzz\n"));`

night-ghost, 2017-04-16:
number of 'z' guessed in 

v930
* fixed startup hang from v929

now this line is 
`Serial.print_P(PSTR("#1zzzzz\n"));`

closed 2017-04-17 by night-ghost