night-ghost/minimosd-extra #189: WP Distance panel does not work

iskess, 2017-04-15:
#### Issue details
When navigating to waypoints in Auto mode, the WP Distance panel does not operate. Only the WP label displays. 

#### Version
r928

#### Board type
generic MinimOSD

#### TLOG and OSD

[2017-04-10 15-32-55 (OSD Test Log) .zip](https://github.com/night-ghost/minimosd-extra/files/923948/2017-04-10.15-32-55.OSD.Test.Log.zip)

night-ghost, 2017-04-16:
.OSD file?

iskess, 2017-04-16:
[X8 Copter NightGhost.zip](https://github.com/night-ghost/minimosd-extra/files/924185/X8.Copter.NightGhost.zip)

night-ghost, 2017-04-17:
what is hardware config?

iskess, 2017-04-20:
Not sure what you are specifically looking for in my hardware config.  
I have a pixhawk connected to a generic MinimOSD using Telem2. When I load the old MinimOSD-Extra r800 the WP Dist panel works fine.

iskess, 2017-04-23:
Is there any other information I can get for you?

iskess, 2017-05-02:
@Night-Ghost,
Is there a chance this bug might be fixed by this weekend? I have a big WP mission planned. I know it's free software so I don't mean to be pushy, if you aren't going to be able to get to it I will just have to load up the old r800 and reconfigure. There are just so many other new features I would be missing. 
Thank you for bringing MinimOSD back to life for the community.

night-ghost, 2017-05-02:
Sure no - my USB hub emitted a magic smoke, new one comes from China by service snails of Russian Post.

night-ghost, 2017-05-02:
BTW, you still not shows photo/video for this issue

iskess, 2017-05-03:
[image]
 Here you go.

night-ghost, 2017-05-10:
should be fixed in v934

night-ghost, 2017-05-19:
no answer for a week - closed

closed 2017-05-19 by night-ghost