night-ghost/minimosd-extra #191: Px4 incompatibility?

4cellOstrich, 2017-04-20:
#### Issue details
Hi, I am having issues with PX4 on pixhawk and your osd firmware, I have tested this on both a traditional minimosd as well as the diatone SP3 with an identical issue. When I Plug in the battery i get all heading information as well as compass readings, however, it always lacks flightmode, as well as gps data. i am plugged into telem 2 and i have tried MANY firmwares all with the same exact issue. Thank you for your time. 

#### Version
many

#### Board type
minim and sp3

#### How "canonical" OSD firmware works?
[  ] Works fine
[  ] The same bug
[X  ] Not applicable
[  ] Not tested

#### TLOG and OSD
_Please provide a links to a relevant screenshot/video, TLOG and OSD files that show the issue_

night-ghost, 2017-04-23:
Yes I don't use PX4 firmware and never  test it with OSD.

4cellOstrich, 2017-05-12:
Would you have any suggestions for trying to force compatibility from what I've been experiencing? Any suggestions on where to look for the issue, it works for 3/4 of the stuff so I assume there is just a minor disconnect in the source somewhere.

tops4u, 2017-05-12:
@4cellOstrich I can not confirm your Observation. I use the OSD FW for over 2 Years now with PX4 and the only Issue Ihave is that the FlightMode always shows Stabilized. Lately I wanted to investigate how to fix this but I have not yet found time to do so.

4cellOstrich, 2017-05-12:
@tops4u That I believe at the moment is my only issue as well, do you think it might be requesting data from the wrong location and that's why it has that issue?

night-ghost, 2017-05-19:
no answer for a week - closed

tops4u, 2017-05-19:
@night-ghost I think it would be nice to include the PX4 Mavlink Flavour (Flightmode Mapping)...

night-ghost, 2017-05-19:
I don't use PX4 firmware stack so I can't test and debug it. If anyone makes PR for this support - welcome.

tops4u, 2017-05-19:
I do some occasional Programming on ATMEL but I'm no expert and I don't know the Mavlink Protocol at all... so this might take some time. But I'll try.

night-ghost, 2017-05-19:
If you can tell me how PX4 sends flight modes and how to distinguish PX4 from others by its mavlink then I can to try to do it.

tops4u, 2017-05-19:
I'll try to get in Contact with some PX4 DEVs.

night-ghost, 2017-05-19:
Ok, flash now has 82bytes free :)

4cellOstrich, 2017-05-19:
I really can't thank you guys enough for this, I've been stuck on this for months, if at any point you guys want more info just ask!

4cellOstrich, 2017-05-23:
I have been researching mavlink and the possible differences between PX4 and ardupilot and at the moment the results are very inconclusive. If you have any new progress please let me know!

tops4u, 2017-05-23:
I think APM and PX4 handle Flightmodes very different. APM uses a simple Enumeration while PX4 has a hierarchy of Modes and Submodes. 

Check out the Plugins that use APM & PX4 in QGroundcontrol.

https://github.com/mavlink/qgroundcontrol/blob/4423b2bf3e7ea4e06458141f692ad3243cd4eff5/src/FirmwarePlugin/APM/ArduCopterFirmwarePlugin.h

Respectively

https://github.com/mavlink/qgroundcontrol/blob/4423b2bf3e7ea4e06458141f692ad3243cd4eff5/src/FirmwarePlugin/PX4/PX4FirmwarePlugin.h

4cellOstrich, 2017-05-24:
I also found today 2 new kinds OSD software, that claimed they supported px4, those being MultiOSD as well as MinimOSD-X2, I attempted to flash X2 and was met with only waiting for mavlink heartbeats, however, the flash may not have been successful as i noticed that my antivirus suppressed certain files, but that was (likely) a false positive, i couldn't find a .hex in the release of Multi, so i couldn't really do anything there. Have either of you heard of these versions? if so any luck?

night-ghost, 2017-05-25:
> if so any luck?

I could try to answer if there were references to the mentioned OSD projects

4cellOstrich, 2017-05-25:
right right of course:

MultiOSD -- https://github.com/UncleRus/MultiOSD

MinimOSD-X2 -- https://github.com/svpcom/MinimOSD-X2

night-ghost, 2017-05-25:
1st.  case MAV_AUTOPILOT_PX4 and follow calls. 

2nd. panFlightMode shows it for px4, case MAVLINK_MSG_ID_HEARTBEAT gets mode. So anyone can do copy&paste if want PX4 support. 

Don't forget for #ifdef's

yashm, 2017-05-25:
Hi @night-ghost, could you please elaborate what you would like from our side?
The OSD support for PX4 would be of great benefit to a large number of us.

Thanks for a great project!

night-ghost, 2017-05-26:
Now I haven't plans to add this support, so there will be anything to test only if someone will make a PR with such changes.

4cellOstrich, 2017-05-26:
@night-ghost could you elaborate what you mean by PR?

night-ghost, 2017-05-26:
do you know how GitHub works?

yashm, 2017-05-26:
@night-ghost do you want me to make a private repo for the px4 development?

night-ghost, 2017-05-26:
>do you want me to make a private repo

you must be joking!  Really, whether I can forbid something? Project is open source so it free for everyone and it covered by GPLv3 license

yashm, 2017-05-26:
>  so there will be anything to test only if someone will make a PR with such changes.

I thought that's what you meant by "PR." I wasn't trying to restrict anything.
What "PR" are you referring to?

night-ghost, 2017-05-26:
We are now in GitHub. Do you read any GitHub documentation? Do you ever listen about Pull Request?

night-ghost, 2017-06-21:
v938
* Preliminary support for PX4 firmware, use MAVLINKPX4 protocol version

closed 2017-06-21 by night-ghost