night-ghost/minimosd-extra #196: Re-built firmware .hex "is too big"

maxim-kuvyrkov, 2017-05-10:
#### Issue details
I'm trying to rebuild minimosd firmware with baud-rate 19200 (using together with DragonLink serial modem, which can't go faster than 19200).  The build finishes fine, with the following size statistic:
===
/usr/share/arduino/hardware/tools/avr/bin/avr-size --mcu=atmega328p -C --format=avr build-atmega328/MinimOsd_Extra.elf
AVR Memory Usage
----------------
Device: atmega328p

Program:   31752 bytes (96.9% Full)
(.text + .data + .bootloader)

Data:       1598 bytes (78.0% Full)
(.data + .bss + .noinit)
===

However when I try to upload firmware via configurator app, it complains "File too big!".  Your released .hex files upload just fine.

I tried with Ubuntu Trusty and Xenial.

Any ideas?  Which distro version / toolchain version are you using?

#### Version
_What version was the issue encountered with_

#### Board type
_What type of the OSD board you are using?_

#### How "canonical" OSD firmware works?
[  ] Works fine
[  ] The same bug
[  ] Not applicable
[  ] Not tested

#### TLOG and OSD
_Please provide a links to a relevant screenshot/video, TLOG and OSD files that show the issue_

night-ghost, 2017-05-11:
This is YOUR hex is too big, not my :)

just comment out some features

night-ghost, 2017-05-11:
and you don't need to recompile to change baudrate

closed 2017-05-11 by night-ghost

maxim-kuvyrkov, 2017-05-11:
То что это /мой/ hex слишком большой -- и ежу понятно.  Хотя размер файла такой же, как и у released hex'ов -- 85kb.  Я полагаю, что Вы делаете release'ы из исходников без изменений, что подтверждает код pack.sh скрипта -- поэтому странно, что нужно что-то отключать, чтобы re-build прошивка заливалась.

Но бог с ним, основной вопрос -- как изменить baudrate?

И, если не сложно, уточните на какими инструментами (arduino version, OS/distro version) вы собираете.

Спасибо.

night-ghost, 2017-05-11:
Собирается все специальным билдом GCC, заточеным под более оптимальный код для AVR, поэтому чтобы лезло после обычного GCC нужно что-нибудь отключить. Нашел где-то на форуме Ардуино, повторно искать не буду.

baudrate не надо менять, работает автоопределение скорости порта больше года как

night-ghost, 2017-05-11:
Для 935 версии

Program:   30516 bytes (93.1% Full)
(.text + .data + .bootloader)

Data:       1603 bytes (78.3% Full)
(.data + .bss + .noinit)

автоопределение будет работать даже если только TX подключен.

maxim-kuvyrkov, 2017-05-11:
Just to report back, autobaud does work even with only one RX line on minimosd side connected.  Don't know why it didn't work for me initially.

The size of firmware built with Ubuntu Xenial gets down to appropriate limits if "#define AUTOBAUD 0" is commented out in Config.h header.

sincoder, 2017-05-14:
@maxim-kuvyrkov did you find the solution ?  how to build a smaller firmware. the smallest firmware i got is 87KB and it still to big to flash

maxim-kuvyrkov, 2017-05-15:
@sincoder , as written above (in Russian), release .hex are built with a custom AVR toolchain optimized for size.  That toolchain was found on some forgotten forum.  I didn't succeed in re-building full version of firmware with Ubuntu-provided toolchains.

What I did is disabled features in Config.h, which allowed firmware to fit.