night-ghost/minimosd-extra #197: how to build the source

sincoder, 2017-05-11:
As i always got error:
**

> protocols/MAVLink.h:79:18: error: ‘class SingleSerial’ has no member named ‘available_S’
>      while(Serial.available_S()) {
>                   ^
> protocols/MAVLink.h:80:28: error: ‘class SingleSerial’ has no member named ‘read_S’
>          uint8_t c = Serial.read_S();
>                             ^

**

i have update the submodule , installed all need packages

maxim-kuvyrkov, 2017-05-11:
What version (MAVLINK, NMEA, etc) are you compiling for?  I found it easiest to trim build.sh to the single version I'm interested in.

sincoder, 2017-05-12:
@maxim-kuvyrkov what i do is:
**

> git clone https://github.com/night-ghost/minimosd-extra.git
> git submodule init
> git submodule update
> ./build.sh   <---- i got the error
**

nothing change, with default config

maxim-kuvyrkov, 2017-05-12:
SingleSerial revision checked out by "git submodule update" is too old; do "cd libraries/SingleSerial; git checkout master" and ./build.sh will succeed (at least for everything but UAVTALK).

sincoder, 2017-05-12:
@maxim-kuvyrkov thanks for your information,I finally did a successful build

closed 2017-05-12 by sincoder