night-ghost/minimosd-extra #202: 933DV-mavlink does not seem to detect landing and show summary page (for plane)

marcmerlin, 2017-06-12:
It didn't work reliably in older firmwares, but sometimes it switched the summary page after I landed and disarmed.
Now it doesn't seem to work at all, unless I hack the code in MinimOsd_Extra/Panels.h
if ( !lflags.motor_armed && landed /* not 0! */ && time_since(&landed) < 3000
I can change this code to trigger the summary page, but then it triggers at the wrong time.

Part of my problem is that I need a clean/simple tlog file with takeoff, 1mn flight, landing, disarm, and then 10sec of recording after disarm to give a chance to the code to kick in and show the summary.
I have not been able to record a clean simple tlog yet, and the tlog player does not allow to skip forward on a long flight I think, so it's annoying to debug with that.
Do you maybe have a simple tlog that's less than 2mn, showing takeoff/landing?

marcmerlin, 2017-06-13:
If that helps I've just made a synthetic plane takeoff/landing flight tlog
It shows takeoff, short flight, landing, "throttle disarmed", and a bit more time.
At no time do I see the summary screen.

[debug_flight.tlog.gz](https://github.com/night-ghost/minimosd-extra/files/1070360/debug_flight.tlog.gz)

marcmerlin, 2017-06-13:
Ok, I found a fix it seems:
        if ( !lflags.motor_armed && landed // && time_since(&landed) < 3000 
          && ((int)trip_distance > 5) 

I had to comment out time_since(&landed) < 3000 
Not sure why it's there or why it's not working, but it's fine without it.
I'm attaching a fixed firmware if that helps anyone (not flight tested yet, but passes my test)
[MinimOsd_Extra_Uni.936MM-MAVLINK-release.hex.gz](https://github.com/night-ghost/minimosd-extra/files/1070401/MinimOsd_Extra_Uni.936MM-MAVLINK-release.hex.gz)

night-ghost, 2017-06-13:
Thanks, fixed in v936

closed 2017-06-13 by night-ghost

marcmerlin, 2017-06-27:
I flight tested this yesterday and I can confirm it works fine on a real flight too. Sorry for the delay

night-ghost, 2017-06-27:
Fine!