night-ghost/minimosd-extra #204: No OSD on Firmwares 920 and above

4cellOstrich, 2017-06-30:
Hello,

First of all, cannot thank you enough for the PX4 compatibility update, that was a lifesaver. Second of all, the issue. I tried flashing 938-MAVPX4 and the flash was successful, however, when I powered the osd, there was no characters displayed, and when I tried to flash from Arduino I only got error after error, i.e. "#include <single.serial>" must be first and such, I worked through a couple of these, either adding in the .h files it couldn't find or adding the .zip libraries it couldn't find either. However, each time I fixed one thing another would come up until eventually I was stuck on it being completely unable to find "mavlink_types.h" in the v1.0 area. Giving up here, I went on to test where the firmwares stopped working and I found that somewhere between 920-923 it stopped displaying characters, I have this margin here because the files between these firmwares were too big to flash.  So that is where I am, I trust that the new update is compatable but I have no way of seeing it right now.

Thanks in advance!!

AndrejSlegl, 2017-07-14:
I have noticed the same issue with minim OSD v1.1

wiatrM, 2017-07-18:
Yes, the same. DO we have any solution for this?

helmarw, 2017-07-26:
same here... i tried 938,937 and 936
didnt feel like going back one by one since i wasnt sure it will work at all, so i took the 880 from stable
and everything is fine.

urlu75, 2017-08-17:
same issue, MAVLink et MAVLink PX4 938,37 and 36 not working.
Last working FW for me is 934 MAVLink.

pwbecker, 2017-08-20:
I have managed to get v938 working by an exhaustive process of comparing v935 (working) and v936 (not working).

I found that the defn of Spi had to be removed from the .ino file, and the extern qualifier removed from Spi.c. Also, to make the code fit, in Config.h I had to comment out DEBUG, USE_SETUP and AUTOBAUD.

As an aside, to get it compiling I had to download SingleSerial, and edit the Makefile to include proper definitions of BOARDS_TXT, ARDUINO_CORE_PATH and  ARDUINO_VAR_PATH. Also, 0-build.sh had to be changed to use MAVLINK instead of MAVLINKPX4, as I am using an APM2.6.

Finally, the include of GCS_MAVLink/include/mavlink/v1.0/mavlink_types.h in protocols.h had to change to the correct path, <include/mavlink/v1.0/mavlink_types.h>.

Hope that helps!

raygamero, 2017-10-25:
Wow @pwbecker, that's some work! I also have the same issue.  Any word on a fix so I can use this with my pixhawk PX4?

Houtkabouter, 2017-11-11:
I along had an issue with no OSD overlay on a Minim Micro. went back to 880 Mavlink and working fine. Thanks so much for all the effort to code this! We're using it to fight farm attacks in South Africa and the OSD data helps with the conviction in court.

night-ghost, 2017-11-13:
Does anyone reads Changelog??? Last working version is 934, there was only changes related to NMEA protocol later

closed 2017-11-13 by night-ghost

pwbecker, 2017-11-15:
@raygamero the fixes needed are described in my previous post.  Most are obvious when you try compile using 0-build.sh, like the wrong pathnames - and those will depend on your system, so what I did may not be right for you. The only tricky one was moving the declaration of Spi from the .ino into Spi.c. Are you able to try that, and recompile?