night-ghost/minimosd-extra #205: Mah used

malpoma, 2017-08-16:
#### Issue details
Hi can i have on OSD the Mah used in flight? not pnly % remaning?
#### Version
_What version was the issue encountered with_

#### Board type
_What type of the OSD board you are using?_

#### How "canonical" OSD firmware works?
[  ] Works fine
[  ] The same bug
[  ] Not applicable
[  ] Not tested

#### TLOG and OSD
_Please provide a links to a relevant screenshot/video, TLOG and OSD files that show the issue_

pwbecker, 2017-08-19:
Yes, just select Battery Percent on one of the panels, then a dialog will be displayed where you can uncheck %

night-ghost, 2017-11-13:
this is not issue but support question and should be in forum

closed 2017-11-13 by night-ghost