night-ghost/minimosd-extra #207: MWII roll and pitch angles * 10

pwbecker, 2017-08-23:
#### Issue details
MWII results in angles being displayed for roll and pitch which are ten times too high

#### Version
938

#### Board type
MinimOSD (329P)

#### How "canonical" OSD firmware works?
[  ] Works fine
[  ] The same bug
[ x] Not applicable
[  ] Not tested

#### TLOG and OSD
-

The fix is in protocols/cleanflight_core.h:
 	osd_att.roll = mwii_read_uint(offsetof(MW_ATTITUDE_t, Angle[0]) );
+        osd_att.roll /= 10; // MSP is centidegrees
 	osd_att.pitch  = mwii_read_uint(offsetof(MW_ATTITUDE_t, Angle[1]) );
+        osd_att.pitch /= 10; // MSP is centidegrees

night-ghost, 2017-11-13:
thanks, will be in next release

night-ghost, 2017-12-21:
done

closed 2017-12-21 by night-ghost