night-ghost/minimosd-extra #208: can you tell me how to build this project?

wetouns, 2017-09-15:
I found bug about radar and I know how to fix it,but I don't know how to build this project and use witch IDE to build,can you tell me?

wetouns, 2017-09-16:
I use Arduino 1.8.4 to build project but the .hex file is 88k,when i use config tool to update fireware it alert "file too big!",what should I do?

night-ghost, 2017-11-13:
report that bug here

pwbecker, 2017-11-13:
It doesn't build with the IDE.  There is a .sh build file that can be used.  It may need to be edited for your system, and you may need to edit out some features from Config.h to reduce the code size

0n3-70uch, 2017-12-19:
Hi, here is a tutorial for building the code with Linux:
https://www.rcgroups.com/forums/showpost.php?p=35359151&postcount=1266

But it's a bit outdated...

night-ghost, 2017-12-21:
I just build via 0-BUILD.sh (CentOS 6.5)

closed 2017-12-21 by night-ghost