night-ghost/minimosd-extra #210: CharSet Uploading

bionicbone, 2017-11-21:
#### Issue details
Failed to upload CharSet many times

#### Version
931 & 933 - MavLink Version

#### Board type
MiniOSD

#### How "canonical" OSD firmware works?
[  ] Works fine
[  ] The same bug
[  ] Not applicable
[X] Not tested

Changed Wiki Page for the CharSet uploading, seems better to use the firmware Character_Updater_FW.hex   -- Not sure what the "Character_Updater_FW_1.hex" firmware is for ???

Raising issue only to ensure you are happy with the WiKi change

Regards
Kev

night-ghost, 2017-11-23:
read docs - https://github.com/night-ghost/minimosd-extra/wiki/Charset

night-ghost, 2017-11-23:
do you read Changelog carefully? 

v925
* font uploader moved out from main firmware. Use Character_Updater_FW.hex instead.

closed 2017-11-23 by night-ghost