night-ghost/minimosd-extra #213: Radar not working

JeyPi, 2018-01-23:
Issue details
The Aircraft symbol on the radar field stays stationary during the flight. It only orients somehow but does not travel. 

#### Version
the latest

Board type
https://www.ebay.com/itm/On-Screen-Display-APM-Minim-OSD-V1-1-Module-for-ArduPilot-Mega-MAVLink-Optimal/322263494662?hash=item4b0866b006:g:bVwAAOSwLF1X25fE

TLOG and OSD
Will do as soon as I get to flying again

night-ghost, 2018-01-23:
tlog and .osd files are required

JeyPi, 2018-02-01:
I noticed that the wind speed and direction indicator is wrong in the mission planner as well. I think it is a mistake of the arduplane software than.
[T_log and OSD.zip](https://github.com/night-ghost/minimosd-extra/files/1685115/T_log.and.OSD.zip)
If you need the video it is big. I can post it on drop box.

JeyPi, 2018-02-11:
I did tried older software and can confirm that the radar feature is not working with 934.
As well I noticed that the home altitude is the same as the flown altitude. I do not know if it is supposed to be. I also would suggest that the radar aircraft symbol would be like others have it. The symbol like this >
It is more obvious and less difficult to recon.

night-ghost, 2018-02-20:
I don't use radar, so I asked those who need it to make a certain flight and record video and tlog. I am still waiting.

JeyPi, 2018-02-26:
25 days ago I posted a T_log and osd.zip
All is there.

mric3412, 2018-05-22:
After flight test, I wouldn't recommand flashing this .hex (cf. last comment) : radar works but the arrow indicating home direction can be wrong (+180degres)

mric3412, 2018-05-25:
I tested radar (for plane) with the last commit (2018/05/21). UAV symbol ant 3 track points are moving but not correctly. It should move around the center of the HUD frame but it's only moving in the top right quarter.
In Func.h@505and506 we have dst_x and dst_y always >0 (fabs())
=> in Panels.h@411and412 whe have x et y always >0 too (since dst_x and dst_y always >0)
=> in Panels.h@432, UAV symbol is always writed at (center_col+(x>0) ,center_line-(y>0)) : so, always at the top right quarter of the HUD frame.
In Panels.h@432, x and y should be positive or negative to have a correct radar.

night-ghost, 2018-05-25:
thanks for investigations, will fix ASAP

JeyPi, 2018-06-05:
I flew the last version and the radar icon moves within the hud I would say correctly. I do not know if that is to be or not but the icon is not rotating according to the heading to home.

mric3412, 2018-09-25:
In v946,
when plane is heading North, icon radar is going down
when plane is heading South, icon radar is going up
I don't know if this convention is the most natural for most people, but I would have chosen the opposite (north up on the screen)
To make the North appear up on the screen, simply modify panels.h@432 :
"+ y" instead of "-y" here :     OSD::write_xy(center_col + x, center_line - y, radar_char());