night-ghost/minimosd-extra #217: Wrong UART speed for MWII protocol

pmarkiewicz, 2018-03-17:
Wrong UART speed for MWII protocol, it's 57600 instead of 115200.
In case of MWII AUTOBAUD shouldn't be used.

Looks like case is not related to current hardware (OZE32 board, Naze32 + minimOSD integrated).

night-ghost, 2018-04-26:
can be MWII protocol on other speed than 115200?

pmarkiewicz, 2018-04-26:
never seen other speed, you can of course compile firmware with different
speed.
THis is from anciet times of MultiiWiiCopter on ATMega.

Also here is hardcoded
https://github.com/ShikOfTheRa/scarab-osd/blob/131ffd412383cd1d9d747b74d5f177d628481504/MW_OSD/Def.h
And in iNav there is:  serialConfig->portConfigs[i].msp_baudrateIndex =
BAUD_115200

For whatever reason autodetection doesn't work in case of MSP (I didn't
investigated this part)

regards

pmarkiewicz, 2018-04-26:
> 131ffd412383cd1d9d747b74d5f177d628481504/MW_OSD/Def.h

but https://github.com/iNavFlight/inav/issues/3127

2018-04-26 13:25 GMT+01:00 Piotr Markiewicz <piotr.markiewicz@gmail.com>:

night-ghost, 2018-04-26:
so it is too early to drop autobaud

pmarkiewicz, 2018-04-26:
probably, let's wait for answer.

pmarkiewicz, 2018-04-27:
We got answer from iNav team and still have no idea why it's not 115.
My problem is that baud autodetection code didn't work for me that why I
disabled one and left hardcoded value.
I think we can keep this hardcoded unless someone will raise an issue
related to that.

2018-04-26 13:41 GMT+01:00 Piotr Markiewicz <piotr.markiewicz@gmail.com>:

night-ghost, 2018-04-27:
the answer was a really cool :)

autobaud was highly tested and works just fine for MAVlink, so it seems very strange to me. But as you wish, it will be turned off for MWII in next version, so please remove this part from PR

pmarkiewicz, 2018-04-27:
anothere answer clarifies a bit.
I think you can reject PR and I will create new one just for roll/pitch and
voltage

BondarenkoA, 2018-06-07:
i tested last commit. MWII uart speed still 57600. Because #define AUTOBAUD 0 in the func.h:getData(). When it commented, MWII stay worked, but uart speed does not dysplayed on the diagnostic screen.

night-ghost, 2018-06-08:
Thanks, that "define" was a bug, fixed. UARD speed shown only if autobaud is turned on. If you can build binaries please uncomment "define DEBUG", turn on autobaud and see to measured pulse time.