night-ghost/minimosd-extra #224: PX4 Messages not shown

tops4u, 2018-05-04:
#### Issue details
MAVLINK Messages are no longer shown. They have started to disappear last year, probably due to a change on the PX4 side. 

#### Version
934 onward

#### Board type
MicroMininmOSD

#### How "canonical" OSD firmware works?
[  ] Works fine
[  ] The same bug
[  ] Not applicable
[X ] Not tested

#### TLOG and OSD
Currently not available, I will produce a short TLOG (which can be done on the ground and flip the "Kill Switch" which will trigger a MAVLINK Warning Message).
I assume the Message is transferred via http://mavlink.org/messages/common#STATUSTEXT

night-ghost, 2018-05-08:
>They have started to disappear last year, probably due to a change on the PX4 side.

you should to open issue on PX4 side because they breaks things

tops4u, 2018-05-25:
Probably this is due to Mavlink v1.0 differs to 0.9 in a way that prior to 1.0 the STATUSTEXT was Message ID 254, whereas in 1.0 it is 253.