night-ghost/minimosd-extra #225: ILS direction

mric3412, 2018-05-07:
ILS for plane is not working when you want to land in the same direction than for takeoff
Planes generaly takes off and lands in the same direction (case of constant wind's direction)
An ILS working in all approach's direction would be a good option in most cases.
VERSIONS : v943 / v942 (older versions not tested)
BOARD TYPE : Minim OSD V1.1 (ATMEGA328 and MAX7456)

night-ghost, 2018-05-08:
required files?

mric3412, 2018-05-08:
It can be tested with the following tlog file
[log.zip](https://github.com/night-ghost/minimosd-extra/files/1984052/log.zip)
Select only the ILS option of the HUD panel to clearly observe the ILS behavior : very hard (impossible?) to get the pair of horizontal and vertical lines (arrows suggesting to take another direction are very rarely absent)

night-ghost, 2018-05-16:
waiting for your PR

mric3412, 2018-05-23:
I tested in flight ILS (for plane) with softer angle constraints to display more easily horizontal and vertical lines of ILS (without softer constraints, these lines were too rarely displayed to be tested). Now it seems to me that ILS disfunctions.
For example, look at the current version of Panels.h (line 305) : if the plane is aligned with the runway, we want vertical lines at the middle of HUD <=> lineposition=totalNumberVertLines/2 <=> currentAngleDisplacement=0. But, because of line 309, 0degres will become -180degres.
In v857, Panels.h@286 currentAngleDisplacement==0 => linePosition==(totalNumberOfLines/2). So it seems OK until v857 but it seem's to me that giving the possibility to use ILS in opposite direction introduced a bug.

night-ghost, 2018-05-23:
Thanks, will see