night-ghost/minimosd-extra #228: FlightMode Displaying Garbage and Not Updating

CLabeck, 2018-05-22:
#### Issue details
The flightmode is not showing correctly.

The flightmode field permanently displays "M_0". 

It does not show a value that corresponds to a real flight mode (e.g. STAB). 
The shown value does not change when the flight mode on ArduCopter changes.

The field is basically statically displaying the wrong content.

All other fields work perfectly!

#### Version
Using 9.43

#### Board type
DARLC minimOSD (https://www.fpv24.com/dalrc/dalrc-minimosd)

#### How "canonical" OSD firmware works?
[  ] Works fine
[  ] The same bug
[  ] Not applicable
[ X ] Not tested (what is canonical?)

#### TLOG and OSD
Screenshot below
[image]

BondarenkoA, 2018-05-23:
Fixed within PR #229

night-ghost, 2018-06-01:
m_0 means that current mode has code 0. What protocol are you use?

closed 2018-06-01 by night-ghost