night-ghost/minimosd-extra #230: Delta Values & Wrong Flightmode

tops4u, 2018-05-25:
#### Issue details
Values like Altitude, Distance, Power Used etc are delta calculated Values instead of decoded Values from the MAVLINK Datastream (where this information is available). 
To proof this, I have a case where there was a short interrupt in the Power Supply (5v) to the OSD during a Mission Flight. 
Check Screenshot just before the Outage:
[image]

And afterwards:
[image]

As you can see several Values (Altitude!!, Distance flown, Distance to Home, Power Used!!, Flight Time,...) have been reset. Just to note that the Flightmode selected was Mission and not RTL. 

#### Version
R934 MAVLINK PX4

#### Board type
MicrominimOSD

#### How "canonical" OSD firmware works?
[  ] Works fine
[  ] The same bug
[X] Not applicable
[  ] Not tested

#### TLOG and OSD

night-ghost, 2018-06-01:
this is feature.

closed 2018-06-01 by night-ghost

tops4u, 2018-06-01:
Please don't let me die stupid... what is the use of this?

night-ghost, 2018-06-01:
MAVlink stream don't contains this data so they are calculated by OSD itself.

tops4u, 2018-06-02:
Not completely true. Altitude (Above Ground) as well as Current Consumed are available in PX4 Mavlink Stream. I concur that Travel Distance, Distance to Home and Home Alt are not in the Mavlink Stream. However IMHO the vital Flight Data (like ALTITUDE! and a little less the Current Consumed) should always be taken from the stream instead of being calculated by the OSD.

night-ghost, 2018-06-02:
Really? you could fix it yourself, PRs are welcome.  Did you read a license?